Sienci-Labs/gsender #593: gSender 1.4.11 - M1 Mac - Server Disconnect Message

cjazz, 2025-01-08:
I found some strange behavior running gSender (1.4.11) on 12/31.

This is a repeat message from gSender Facebook group.

Info:
• gSender 1.4.11 Universal for Mac
• Running an M1 MacBookPro / macOS Sequoia 15.2
• SienciLabs - LongMill MKII - regular controller (non SLB).
I ran it on a test piece of MDF - An operation for milling guitar fret slots.    It's a 30 minute job, and verified ok.   I went about my business in the shop doing some other orgnizational things.  I noticed the gSender screen would flash quickly.  
So fast I could barely see what it was saying.. So I recorded video of it, and freeze framed on it.

See attached video & Pic - a bit blurry but a "Server Disconnected" message.

The job finished fine, but there's definitely some interoperability issue /glitch between the software and controller.

[image]

https://github.com/user-attachments/assets/64e782cb-f5f6-4f79-a722-bad460379f3c

Note: I have since switched back to a prior version for the time being as I am not confident I won't run into an issue with this behavior yet.

If you need more info or would like me to try something else, please let me know.
Adam

cjazz, 2025-01-14:
crickets

hamanjam, 2025-01-14:
Almost impossible to diagnose.  First guess would be some type of static issue since most places are drier this time of year.  Most people have tried other cables, ports or at least saved a diag file to look at

cjazz, 2025-01-14:
@hamanjam - Thanks for the feedback.   I'll look into static electricity on my end (cables/grounding, etc..)  

Where does gSender store diag files so I can either share or look at myself.   Since this behavior sends those alerts and goes way I'm assuming we've reached a point where gSender hits a retry.

cjazz, 2025-01-16:
Hi @hamanjam

Yesterday I ran with the older version of gSender 1.4.9, and wasn't experiencing the server disconnects (attached a diagnostics).   I did notice that when switching to a browser (Safari), and coming back to gSender, I would find gSender disconnected.   

Note: When I switch to another app, I'm never doing so while gSender is running a job

[diagnostics_1-15-2025_16-53-09.pdf](https://github.com/user-attachments/files/18437396/diagnostics_1-15-2025_16-53-09.pdf)

I'll run another test today with gSender 1.4.11

cjazz, 2025-01-26:
I'm closing this ticket as I believe it was caused by a dust collector grounding or static issue in my shop.

closed 2025-01-26 by cjazz