Sienci-Labs/gsender #596: Webhook support for program events

tillig, 2025-01-21:
There are gcode entries for program events - start, stop, etc. but there's no way to integrate with an alert on those events, like an IFTTT webhook to alert me on program completion. 

It'd be nice to have webhook support for program events so it can be integrated with other systems.

darickc, 2025-06-04:
I created an application that will connect to gsender and publishes all the events to MQTT. You can run it locally, but made more to run in a docker container.  I could modify it to add webhook capabilites if that is something that might be wanted.  I'm going to be using it to hook into home assistant to turn on/off my dust collection when running a job.  Check it out here: [https://github.com/darickc/gsender2mqtt](https://github.com/darickc/gsender2mqtt).

darickc, 2025-08-14:
@SienciLabs No.  Maybe it is whitewolf from this thread: https://forum.sienci.com/t/homeassistant-integration/14111.