Sienci-Labs/gsender #597: Tool change problem - V1.4.11

alexrstephens, 2025-02-06:
I ran into an issue with the tool change operation after updating to 1.4.11, I'll try to specify it below.

I have a fixed tool sensor and have my machine set up using the 'Code' tool change strategy which was working very nicely.  I installed 1.4.11 this morning and tool changes stopped working correctly.

When I would start a new job it would run the 'before' code block, moving the spindle above my tool sensor and stop.  This is normally where it should pop up a dialog asking to change to the new tool and press resume, but that dialog never came up.  I verified in the console it ran the entirety of the 'before' code block.

If I click the Resume Job button in the corner it completely skips the 'after' code block and goes on to try and run the job with whatever tool is in there.  Interestingly, if I turn on the new Skip Dialog option it does properly run through both 'before' and 'after' blocks correctly though obviously I don't have time to change out the tool without the pause.  If I turn Skip Dialog back off and it goes back to just hanging after the 'before' block.  

It also started giving me errors on larger tool numbers.  A file with an M6 T1 was ok, but an M6 T3005 threw an Error 38 on that gcode line number and stopped the job.  

I reverted back to 1.4.10 and everything is back to running perfectly.

Let me know if you need any additional info.

SienciLabs, 2025-02-28:
@kglovern this might be an instance of the updates we made causing a reversion in behaviour. @alexrstephens if you have the chance to re-create this and could generate a Diagnostic ZIP afterwards then the added files could help us track down what's going on for you

kglovern, 2025-03-11:
We found and fixed an issue with the dialog not appearing when using the grbl controller.

If you are using grbl, be aware that the max T value is 255, T3005 will always throw an error.

alexrstephens, 2025-03-11:
I'll download 1.4.12 and post back the results when I get some time.

Regarding the tool change number, it does work >255 (in this setup) to indicate the tool number in the pop up box:

[image]

[image]

I'm guessing the actual tool change command isn't being passed on to grbl when using the Code strategy, which is why there's no error?

kglovern, 2025-03-12:
You can see it's being commented out - it's not being passed.  The T command is stored in a byte on the firmware so the maximum value is 255.  You can verify this yourself by sending T commands in the console.

[image]

closed 2025-03-12 by kglovern