Sienci-Labs/gsender #600: When building the package I get an error

khvalera, 2025-02-11:
While creating the package I started getting the error:

```
# yarn build-latest

[build-prod-app] (node:35861) DeprecationWarning: Tapable.plugin is deprecated. Use new API on `.hooks` instead
[build-prod-server] error: API request failed
[build-prod-server] 
[build-prod-server] Caused by:
[build-prod-server]     [6] Could not resolve hostname (Could not resolve host: sentry.io)
[build-prod-server] 
[build-prod-server] Add --log-level=[info|debug] or export SENTRY_LOG_LEVEL=[info|debug] to see more output.
[build-prod-server] Please attach the full debug log to all bug reports.
[build-prod-server] node:internal/errors:983
[build-prod-server]   const err = new Error(message);
[build-prod-server]               ^
[build-prod-server] 
[build-prod-server] Error: Command failed: /snug/builds-pkg/gsender/src/gsender-1.4.11/node_modules/@sentry/cli-linux-x64/bin/sentry-cli releases new 359d9bbe7b9372dd9770ca6ff662477f0a7e989b
[build-prod-server] error: API request failed
[build-prod-server] 
[build-prod-server] Caused by:
[build-prod-server]     [6] Could not resolve hostname (Could not resolve host: sentry.io)
[build-prod-server] 
[build-prod-server] Add --log-level=[info|debug] or export SENTRY_LOG_LEVEL=[info|debug] to see more output.
[build-prod-server] Please attach the full debug log to all bug reports.
[build-prod-server] 
[build-prod-server]     at genericNodeError (node:internal/errors:983:15)
[build-prod-server]     at wrappedFn (node:internal/errors:537:14)
[build-prod-server]     at ChildProcess.exithandler (node:child_process:414:12)
[build-prod-server]     at ChildProcess.emit (node:events:507:28)
[build-prod-server]     at maybeClose (node:internal/child_process:1101:16)
[build-prod-server]     at Socket.<anonymous> (node:internal/child_process:457:11)
[build-prod-server]     at Socket.emit (node:events:507:28)
[build-prod-server]     at Pipe.<anonymous> (node:net:351:12) {
[build-prod-server]   code: 1,
[build-prod-server]   killed: false,
[build-prod-server]   signal: null,
[build-prod-server]   cmd: '/snug/builds-pkg/gsender/src/gsender-1.4.11/node_modules/@sentry/cli-linux-x64/bin/sentry-cli releases new 359d9bbe7b9372dd9770ca6ff662477f0a7e989b'
[build-prod-server] }
[build-prod-server] 
```
What could be the problem?

J-eremy, 2025-02-13:
@khvalera 
Your network wide ad blocker is blocking sentry.io, as it should be but to build gsender it "phones home" for some reason and if it cant resolve the domain name it fails.

I ran into this same issue and it left a sour taste in my mouth but to work around it just allow it for the build time and block it again.

Hope this helps.

I dont see why it needs to contact sentry.io at all during the build process and in my opinion this needs to be looked into. I know sentry is used on a bajillion pieces of software but in my opinion, and lots of others including the privacy blocking lists consider it malware.

khvalera, 2025-02-14:
@J-eremy 
Thanks for the answer! 
I didn't understand what exactly was interfering, but the package was still collected, but on another server where there is no anti-malware protection on the gateway :)

J-eremy, 2025-02-14:
> [@J-eremy](https://github.com/J-eremy) Thanks for the answer! I didn't understand what exactly was interfering, but the package was still collected, but on another server where there is no anti-malware protection on the gateway :)

See if you can go to the sentry.io website on the machine your trying to build on. My guess is its being blocked somewhere.

Sentry is a powerful tool for developers but definitely walks that fine line of reporting things automatically without being fully transparent on what, and when it does that.

@kglovern I think it would be a good idea to disable Sentry on builds that arent done by you guys specifically. A simple check could probably be put in to skip the whole thing if the environment variable for your API key isnt available. A lot of block lists now block sentry by default now even at their lowest level of protection, and it causes among other things build issues.

khvalera, 2025-02-17:
> See if you can go to the sentry.io website on the machine your trying to build on. My guess is its being blocked somewhere.
 
Access to the sentry.io site exists, I checked it right away. Later I will try to study the problem in more detail.

SienciLabs, 2025-07-18:
FYI @khvalera @J-eremy sorry for the sentry inconvenience on building. It's possible we set it up in a way that makes independent building harder, so we can take another look when we get the chance to reconfigure it to make community builds easier