Sienci-Labs/gsender #601: [REQUEST] simple way to send external macros to the API for more automation possibilities

J-eremy, 2025-02-13:
I was just wondering if anyone had any more info on the possibility for direct API access to send gcode TO gsender to then be sent to the machine for external control of some automation. The ability is "technically" there because of cncjs but I honestly have zero idea on how to access it, nor is it documented. External pendants were the use for them previously.

Having that ability opens up a world of external controls, buttons, touchscreen UI's.

The edge version is really looking great. Keep up the amazing work.

terjeio, 2025-03-01:
If you have a grblHAL controller such as a SLB with SD card and [expression support](https://github.com/grblHAL/core/wiki/Expressions-and-flow-control) enabled you can create pretty advanced macros that are tightly coupled to controller statuses like last probing result, current position etc. They can be run from the SD card either by `G65` or as named `O` macros.

J-eremy, 2025-03-01:
I do not have a HAL controller because I made the mistake of buying from
carbide3d, so until this controller burns up from heat, and that shouldn't
take long because it's well over it's heat dissipation  budget I'm locked
in lol. Thanks for the reply tho.

Direct api access would be awesome even with that feature grblhal works tho.

closed 2026-04-06 by chrismakesstuf