Sienci-Labs/gsender #602: Any documentation on build environment?

riftcodercode, 2025-02-18:
I'd love to contribute to this project .. once I can get it to build. I saw mentioned in another thread to use node 18.x so I used nvm to switch to that version but I get a plethora of npm errors. I'd like to ensure I'm at least using the same build environment as others. Also, has anyone downloaded the source and successfully run npm install on the current package config? If there are any docs for contributors to point me in the right direction please let me know, I've been unable to find anything on github or the "official documentation"

SienciLabs, 2025-03-01:
Yeah sorry @riftcodercode, we've been busy redoing much of the codebase for the new Edge release (it'll vary in its build process compared to the current main) and never got around to documenting the original environment setup.

Glad to have anyone else jump in if that can offer advice to get you started for the time being - otherwise we should have something available for docs in the next month or so. I unfortunately can't lead you in any good direction since I manage the project but don't know the intricacies of code base setup - that's why I've been tagging issues submissions related to building so they're all grouped to answer them more quickly

J-eremy, 2025-03-01:
@riftcodercode I answered a similar question in #557. I built a Dockerfile in there that gets the job done. Take a look at it and see if it solves any of your problems. It should at the minimum give you an idea of how to get it to build successfully.

riftcodercode, 2025-03-19:
> [@riftcodercode](https://github.com/riftcodercode) I answered a similar question in [#557](https://github.com/Sienci-Labs/gsender/issues/557). I built a Dockerfile in there that gets the job done. [...]

Thanks very much! I'm just getting back to looking at this after I saw the video on the edge release.

I'd prefer to work with that release since it will be the future, and it sounded like perhaps some cleanup being done on the codebase.

riftcodercode, 2025-03-19:
> Yeah sorry [@riftcodercode](https://github.com/riftcodercode), we've been busy redoing much of the codebase for the new Edge release (it'll vary in its build process compared to the current main) [...]

Thanks for the reply. I just watched the youtube video on the edge release update and that got me looking back at this. Is the edge release in the dev branch? I didn't see a branch with that label, but it looks like dev is most active as of late. Just thinking I should probably be looking at that as the future version .. if I am going to be able to contrib something that would be useful in the future. Thanks!

closed 2025-11-10 by SienciLabs