Sienci-Labs/gsender #604: gsender 1.4.11 zero all issues

2Tall2024, 2025-02-23:
my gsender in the gcontrol panel does not zero out the X, Y and Z on the display 
i hit zero all and the values do not change 

also the feed rate override do not function.

is there something in the settings i need to do to fix this?

2Tall

thank you,

2Tall2024, 2025-02-23:
my version of gsender is 1.4.11

kglovern, 2025-02-26:
Your $10 is misconfigured and missing WCS reporting.  Restore it to the default value.  This is a firmware configuration issue, not a gSender issue.

2Tall2024, 2025-02-26:
Oh wow
   
  
   thank you for your help
   
  
   i will look into this when I get back to the machine 
   
  
    
   
  
   i noticed when I fired up the Gcontrol for the first time it configured its self for the new spindle and didn’t have to change anything 
   
  
    
   
  
   thank you