Sienci-Labs/gsender #607: Can not start gSender on Linux Ubuntu

breiler, 2025-03-15:
Congrats on the new release 1.5.0, looks really nice in the screen shots (I am a bit envious)!

I get this error when trying to start it on Ubuntu Linux:
```bash
joacim@kitty:~$ gsender 
LaunchProcess: failed to execvp:
/opt/gSender
[96483:0315/095541.636173:FATAL:zygote_host_impl_linux.cc(207)] Check failed: . : Invalid argument (22)
Trace/breakpoint trap (core dumped)
```

I tried installing it with:
```bash
sudo apt remove --purge gsender
rm -r ~/.config/gSender/
sudo dpkg -i Downloads/gSender-1.5.0-Edge-5-amd64.deb
```

Running Linux Ubuntu:
```
joacim@kitty:~$ uname -a
Linux kitty 6.8.0-55-generic #57-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 12 23:42:21 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux

joacim@kitty:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
```

misnomer, 2025-03-28:
Hi, I was able to get it working by getting rid of the space in the installation directory and adding setuid to the chrome-sandbox executable inside the installed package directory:

```
sudo mv /opt/gSender\ Edge/ /opt/gSender_Edge
sudo chmod 4755 /opt/gSender_Edge/chrome-sandbox
```

I'm using the same distro as above.

Re the space in the directory name, here's a line from strace that shows that whoever is executing the chrome sandbox doesn't appear to be quoting properly:

```
[pid 35149] execve("/opt/gSender", ["/opt/gSender", "Edge/chrome-sandbox", "/opt/gSender Edge/gsender", "--type=zygote"], 0xfd8000cb000 /* 49 vars */) = -1 ENOENT (No such file or directory)
```

Re the setuid, I saw that main.js is adding --no-sandbox to the chromium command line, but main.js itself appears to execute only after this error happens.  

More info re setuid and electron here:

https://github.com/electron/electron/issues/42510

Lastly, when building from source, scripts/electron-build.sh executes the electron binary in the node_modules directory as part of the build:linux yarn script.  This also fails, though it doesn't stop the build and doesn't appear to affect anything.  For example, from a recent github workflow:

https://github.com/Sienci-Labs/gsender/actions/runs/14135190087/job/39605111678#logs

```
$ bash -c 'scripts/electron-builder.sh --linux --x64 --arm64'
[2[9](https://github.com/Sienci-Labs/gsender/actions/runs/14135190087/job/39605111678#step:9:10)
13:0328/181154.964249:FATAL:setuid_sandbox_host.cc(163)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /home/runner/work/gsender/gsender/node_modules/electron/dist/chrome-sandbox is owned by root and has mode 4755.
/home/runner/work/gsender/gsender/node_modules/electron/dist/electron exited with signal SIGTRAP
```

I was unable to get this part working with the chown and chmod route locally- the only thing that worked was the sysctl mentioned in the electron bug report above.  It's a bit of a clobber approach but might come in handy if the maintainers determine that it needs to work for the build to be successful:

```
sysctl -w kernel.apparmor_restrict_unprivileged_userns=0
```

breiler, 2025-07-13:
The latest version 1.5.1 works fine for me now, closing.

closed 2025-07-13 by breiler