Sienci-Labs/gsender #610: [BUG] Version 1.4.12 When the program runs for 3 minutes, it will be a white screen?

Witawat, 2025-03-19:
[image]

Version 1.4.12 When the program runs for 3 minutes, it will be a white screen?

How to fix ?

hamanjam, 2025-03-19:
What OS? Are you connected to a machine and starting a job?   Almost sounds like an old OpenGL issue with Raspberry Pi from the forums but with limited information to go on, it's hard to tell

Witawat, 2025-03-20:
> What OS? Are you connected to a machine and starting a job? Almost sounds like an old OpenGL issue with Raspberry Pi from the forums but with limited information to go on, it's hard to tell

Windows 11 64bit 24h2

connect to cnc and  program runs for 3 minutes, it will be a white screen

i not sure can fix ?

stefan63, 2025-03-21:
Having the same issue on Linux Mint 21.3 Cinnamon  White screen with latest 1.5edge reboot and re-install does not solve the problem :(

stefan63, 2025-03-23:
On LinuxMint 22.11 same solution as "ImagineerNL" used for Windows also works om Mint.
Uninstall gSender edge -> install gSender 1.4.12, -> install/upgrade to gSender edge 1.5 -> now it works (part from other reported bugs). Looking good for the future!