Sienci-Labs/gsender #611: Wrong time estimation when reducing speed

Brexei, 2025-03-20:
I am running gsender (latest version 1.4.12) on my custom made cnc. However, the remaining time estimation does not take into account the speed override settings at all! 
When you run a job and decide that you run too fast and reduce speed (feeds) to 50%, the time to end estimation does not change at all. Then, when you are at nearly 50% done the time estimation is already zero!
Older versions took into account the lines remaining, so the time estimation was changed when a new speed was set.

Brexei, 2025-04-01:
[image]

I am attaching a photo to show that if you change the speed settings even for a couple of minutes then the time estimation is completely wrong!! 
In this example i started the job with 40% speed, just to be safe and after 30sec i increased it to 80%. Then after a minute i set it to 100%. But the time estimation was way off. It does not take into account the lines remaining, nor the speed factor. So, it is clearly a bug.

Brexei, 2025-05-05:
I made some more tests to see what is wrong. It appears that if you do not use the slider then there is a bug and time estimation is wrong. But if you always move it by the slider it gets adjusted.
However, there is another problem. If you have short movements then time estimation is completely wrong!!
Without touching anything at 100% speed, it initially said 35 minutes time estimation. This remained almost there many minuted later. Finally it took 2 hours for the job to finish! So, please review time estimation again. It is much worse than in previous versions. Thank you.

SienciLabs, 2025-07-08:
Thanks for the feedback @Brexei I'm going to turn this issue into just a bulk issue for time estimation in general since it's continued to be a really tricky thing while we've continued to try to make it better, but I also agree it needs more work/fixing so it can become far more reliable for use as a point of reference otherwise people might feel they can't trust it.

Similarly reported issues: #648, #665

Brexei, 2026-02-06:
Thanks. I will test it and see if it works now.