Sienci-Labs/gsender #621: gSender Edge - Automation "File stop" event not executing

siganberg, 2025-04-07:
The "File stop" automation event does not appear to trigger when a file is stopped manually or completes its execution.

**Steps to Reproduce:**

- Open gSender and load a G-code file.
- Set up an automation under Settings > Automation with the "File stop" event.
- Assign any test command or script to this event (move to parking position).
- Start running the file.
- Stop the file manually or let it complete.

**Expected Behavior:**

The automation assigned to the "File stop" event should be triggered as defined.

**Actual Behavior:**
No action occurs — the automation is not executed.

**System Info:**

gSender Version: v1.5.0-Edge-6 
OS: Raspberry Pi OS 64-bit  (self-compiled from original repo)
Hardware: SLB-EXT

kglovern, 2025-04-10:
None of the events are firing - this should be fixed in the next release (it's pushed to dev now if you want to verify)

siganberg, 2025-04-12:
Verified on dev branch on  compiled Pi binaries. @kglovern I will close the issue once it made on the official release.

siganberg, 2025-04-30:
Fixed verified on the new edge version.

closed 2025-04-30 by siganberg