Sienci-Labs/gsender #622: gSender Edge - (Feature Request) Display current tool number on UI (from _current_tool grblHAL variable)

siganberg, 2025-04-07:
**Description:**

It would be great to have the current tool number loaded on the spindle displayed directly in the gSender UI. This small addition would improve workflow visibility, especially for those using tool changers or running tool-specific operations.

Currently, I have to manually enter the following command in the gSender console to see the active tool:
```
$#=_current_tool
```

This value can be retrieved from the grblHAL predefined variables. More specifically, the _current_tool variable: 📘 Reference: grblHAL Expressions and Flow Control - Parameters & Variables

**Suggested Implementation:**

Poll or read the _current_tool variable and display it in a visible section of the UI (such as the header, tool info panel, or under machine status).
Optionally, allow this value to auto-refresh during tool changes or on command send.

Thanks for considering!

kglovern, 2025-04-09:
This is planned for the future as part of a larger update to support ATC.  We'll see if we can get this part in earlier.

siganberg, 2025-04-11:
Just simple display that read that poll _current_tool value should good start before going to big ATC implementation.

terjeio, 2025-04-14:
The current tool is also available in the real-time report, I use that to keep the ioSender UI updated.

siganberg, 2025-08-28:
I just implemented it with my mod version:

https://github.com/siganberg/gsender/releases/tag/v1.5.3-c