Sienci-Labs/gsender #625: G Sender stopping mid carve

Boston-13, 2025-04-10:
I am working on a project carving multiple bottle openers and it does the inside carve ok but when it goes to contour the outline it stops mid carve and then I have to re connect re home and start over. If I try and start from the last line it does that but it moves real slow like 1 inch per minute.
Any direction will be appreciated.

Thank You

tcurdt, 2025-04-14:
Does this here sound similar?

https://github.com/Sienci-Labs/gsender/issues/512

SienciLabs, 2025-07-16:
@Boston-13 at the very least I can say that if you update to any of the new 1.5.# versions the Start from line feature is now fixed. The issue with this was that for imperial files the inch was accidentally being interpreted as mm for feed rate and this was causing those slow movements

Regarding jobs stopping we're still not sure what the cause was in 1.4.# and are now waiting to hear is this is still an issue in 1.5.# and hopefully it's either gone now or we'll have to keep looking into the cause

SienciLabs, 2025-07-16:
Merged #639 over to here.

@Avery63 wrote:

> I'm running a MK2 with a Makita router. Trying to cut a 3D clip art. It starts cutting good then just freezes. Tried a number of times but still stops and not always on the same line. Some times 30 seconds into the cut, sometime half way through the cut. Could it be my Longboard Controller?

@tcurdt wrote:
> Sounds like...
> 
> https://github.com/Sienci-Labs/gsender/issues/513
> https://github.com/Sienci-Labs/gsender/issues/512

SienciLabs, 2025-07-17:
Just to note here, @Avery63 responded in #639 that the source of the file stopping was a bad file and that it hasn't happened since

tcurdt, 2025-08-11:
@walidkayhan I wanted to try and test the most recent version but ran into so many problems with 1.5 that I had to give up/postpone.

https://github.com/Sienci-Labs/gsender/issues/703

UGS works with no problems, gSender 1.5 seems non-functional.