Sienci-Labs/gsender #628: gSender Edge - Suggestion: Vertical Layout Improvements (Jog Area & Panel Spacing)

siganberg, 2025-04-11:
[image]

The current vertical layout in gSender Edge could benefit from some refinements to improve usability and visual balance. Specifically:

**Issues Observed:**

Jog Area has a large amount of unused vertical space.
Bottom panel (including Feedrate Override, Probe, Macros, etc.) feels cramped and requires horizontal scrolling.

**Suggested Improvements:**

Reposition the Feedrate Control Override to better utilize the jog area space.
Expand the width of the "Probe / Macro / Toolchange / etc." panel so that its content fits without the need to scroll.

**Benefits:**

Cleaner and more balanced UI
Improved usability, especially on smaller screens or touch interfaces
More intuitive workflow without constant scrolling or cramped controls

siganberg, 2025-04-12:
FYI on Masso Touch, there is settingns to flip the screen orientation, not sure if this is needed on gSender if you can simply detect the full screen and calculate the screen ratio if its vertical or horizontal layout and make the necessary adjustment.

SienciLabs, 2025-04-12:
This is a goal we've had for this Edge cycle, we just haven't gotten to it yet. You should see it release soon, this is a mockup we made a while back for reference

[image]

kglovern, 2025-04-25:
Dev should have the first iteration of portrait mode now, which will be included in the next Edge release.

siganberg, 2025-04-30:
New layout for portraits is look very good.

closed 2025-04-30 by siganberg