Sienci-Labs/gsender #629: gSender Edge  - Custom Diameter Tool for Probing Not Persistent + Not Sorted in Tool List

siganberg, 2025-04-12:
When adding a custom diameter tool for probing in gSender, the value does not persist after restarting. Each time I reopen gSender, the custom diameter I previously added is lost and has to be re-entered manually.

Additionally, custom tools are not properly sorted in the list of tools. 

**Steps to Reproduce:**

Add a custom diameter tool under the probing settings.
Observe the tool list sorting — custom entries are not sorted correctly.
Close and restart gSender.
Check the probing tool list — the custom tool is gone.

**Expected Behavior:**

Custom probing tools should persist after restarting gSender.
Tool list should sort entries correctly, including any custom-added tools.

**Environment:**

gSender version:  Edge from the recent dev branch
OS: Pi

kglovern, 2025-04-25:
Most of these issues should be fixed in the next Edge release (not currently in dev but should be over the weekend after a bit more testing)

siganberg, 2025-04-30:
Custome diameter still not persistent. When I close the gSender and re-open, the custom diameter that I added is gone.

siganberg, 2025-06-19:
Verified its working now on 1.5.0