Sienci-Labs/gsender #630: gSender Edge  - Unable to Manually Set Travel in XY Below 1 — No Decimal Support

siganberg, 2025-04-13:
When trying to manually change the travel distance in the XY direction via the UI, the input does not accept values below 1. Decimal values (e.g., 0.5 or 0.1) are not working.

Currently, the only workaround is to change the default values for Rapid, Normal, and Precise travel settings in the configuration.

**Steps to Reproduce:**

Go to the Jog Control panel.

Attempt to enter a decimal value (e.g., 0.5) in the XY travel input.

Observe that it doesn’t accept decimal character. 

**Expected Behavior:**

Decimal values for XY travel should be accepted and applied for more precise manual control.

**Environment:**

Version: gSender Edge

kglovern, 2025-04-25:
Should be fixed in next Edge release, changes present in Dev branch

siganberg, 2025-04-30:
Fixed verified.

closed 2025-04-30 by siganberg