Sienci-Labs/gsender #632: M0/M1 should print message

tcurdt, 2025-04-14:
When I have this gcode

```
M0 (please change to 8mm endmill)
```

I would expect the message to show on screen.
Right now it just says "hold" and I need to check the console for details.

SienciLabs, 2025-07-18:
@tcurdt I spoke to @kglovern and he reminded me that we originally supported bracketed comments but were finding that some CNC firmwares were not enjoying them and instead were throwing errors. In light of this, we made the choice to parse out bracketed comments and not send them in the sender to avoid a whole group of people having this issue, and that's why bracketed comments don't work in gSender anymore. That said, the workaround is straightforward which is to use the much more common semicolon comment and it'll show up
```
M0 ;please change to 8mm end mill
```

tcurdt, 2025-07-18:
Perfect. I'll give that a try and then close the issue. Thanks for the follow-up!

SienciLabs, 2025-08-05:
I'll close this in the meantime but am happy to re-open if you have more to report back