Sienci-Labs/gsender #635: gSender Edge - Hide A-axis controls when 4-axis or Rotary is not enabled

siganberg, 2025-04-15:
If 4-axis or Rotary is not enabled in the Rotary Tab or Settings, all A-axis controls should be hidden from the UI. Currently, these controls are still visible even though they have no function unless Rotary is enabled.

This can be confusing for users who don’t use a 4th axis or rotary setup, as it introduces unnecessary UI elements and raises questions about functionality that doesn't apply to their machine.

**Suggested Behavior:**

When Rotary mode is disabled in settings:

- Hide all A-axis controls in the UI.

When Rotary mode is enabled:

- Show A-axis controls as usual.

kglovern, 2025-04-25:
This should be addressed in the next Edge release - A axis functionality will be tied to Config toggle instead of whether the firmware reports that axis.

siganberg, 2025-04-30:
Verified no longer show on the latest edge.

closed 2025-04-30 by siganberg