Sienci-Labs/gsender #636: Gsender not working properly

TYbarrondo, 2025-04-24:
I have set my firmware setting $13 to inches yet jogging moves super slow and it will not stop after I let go of the mouse button, I have to e-stop every time. I have a genmitsu 4040 pro using vcarve pro. Please help I am a total noob to cnc and doa before I even get started. I also set my vcarve to inches and I am using the latest version of both.

rljaffe, 2025-08-01:
I just installed the latest version and when attempting to calibrate x, I submitted my distance in inches and was requested to submit the movement in MM. After converting and submitting the movement, the x travel speed slowed down to a snail's crawl. even on high setting. I had to go back into settings and select defaults. for the operation to go back to normal.
I am still unable to check calibration.

SienciLabs, 2025-08-01:
@rljaffe this is a completely different issue from what @TYbarrondo so it would've been better for you to make a new issue instead, but what you're reporting is a known bug still in 1.5.2 but should be finally fixed in 1.5.3 which we aim to release next Tuesday 👍 so just keep an eye out and let us know how it goes for you

rljaffe, 2025-08-02:
Thanks, will do.