Sienci-Labs/gsender #639: Gsender stopping mid cut.

Avery63, 2025-05-01:
I'm running a MK2 with a Makita router. Trying to cut a 3D clip art. It starts cutting good then just freezes. Tried a number of times but still stops and not always on the same line. Some times 30 seconds into the cut, sometime half way through the cut. Could it be my Longboard Controller?

tcurdt, 2025-05-19:
Sounds like...

https://github.com/Sienci-Labs/gsender/issues/513
https://github.com/Sienci-Labs/gsender/issues/512

tcurdt, 2025-07-17:
Well, using a different file might just hide the problem.

In regards to #512 the gcode sure changes how the machine can work through the buffers. I have the feeling that roundings that don't use a dedicated ARC command (but interpolate) increase the probability for this problem to be encountered.