Sienci-Labs/gsender #646: gSender 1.4.12 and Edge 8 have blank history/job stats

hamanjam, 2025-05-19:
I was answering a Facebook question about run time and when I opened gSender to take a screenshot, my history was blank.  I flipped over to Edge and it was blank too.  

I run only on Ethernet so I'm not sure if that has any bearing but I'm not sure when history disappeared.  If I check the .rc file, there's nothing in there for job history.  I used to use it for estimating jobs based on previous timed jobs but it's been a while since I checked.

Captzer0, 2025-06-02:
I have the same problem. My gSender no longer displays any information regarding Job History & Stats. Overall Stats, Jobs Per Com Port or Run Time Per Com Port are all blank.

SienciLabs, 2025-07-08:
This might've been a bug that wiped these settings during Edge? I can say for certain. Is the issue still present or has the job history gone back to accumulating since you've updated to version 1.5.0 and beyond?

Long-term, we'd like to move tracking job stats to the machines themselves to avoid this issue and also make a better experience by having access to run stats no matter which computer you connect to the machine with

closed 2025-08-04 by kglovern