Sienci-Labs/gsender #654: Gsender Edge 8

itsnotlevel1, 2025-05-31:
I have been using gsender since I got my SLB and I am completely happy. BUT, I was trying to use edge on a linux machine and my mac and neither one will connect to usb. I posted an issue with linux not connecting... after forgetting to add my user to dialup... still, no connect. I then decided to try the latest edge on my mac, which I have been using the current version on the same mac, no issues and an older model of edge and no issues. I downloaded the latest edge and now.... no connection. 
I have unplugged, reset, restarted...machine and mac and nothing... still no connect.

itsnotlevel1, 2025-05-31:
Oh and just tried Edge 7 and it connects.

itsnotlevel1, 2025-07-07:
Just tried the latest ver. 1.5.1 and it installs, opens and hangs on Connecting.... 1.5.0 that I reported before also never connected. Just said.... "connecting" 

1.4.12 is the last version I am able to open... Im on a Mac.

[image]

itsnotlevel1, 2025-07-08:
Sorry, if I need to make a new post I can. Not trying to convolute the post or make it difficult. I have decided that I dont need the bleeding edge stuff, never have. So using the current version gsender...that works... and not using edge is what I am currently focused on. Sorry for high jacking my own post. lol.

walidkayhan, 2025-07-08:
No worries, when you say current version of gSender, are you referring to v.1.4.11, before the big update?

itsnotlevel1, 2025-07-09:
> I'm using 1.4.12 currently bc 1.5.1 wouldn't connect to my Mac. Even tho.. 1.5.1 will connect to my slb via Windows and Linux.
> I'm my hastey attempt to try a fix... And being a Linux user, I tried adding my user to dialout group on the mac and now I have made the usb port unrecognizable to 1.4.12 so now I'm having to work backwards slightly. Not sure how to fix it. I immediately -deleted my user from the group but it didn't fix it. I'll have to see if anyone else has done that and had to correct... Maybe I can find a solution.
>
> Paul
>
> Sent from Proton Mail Android

Ok, fixed. Not sure why the Baud rate changed to zero....I have never changed any of that. 1.4.12 back in service.

itsnotlevel1, 2025-07-10:
Oh and is there any difference in the user folder, application support gsender folder from 1.14.12 to 1.5.1? If so, how do I transfer my config settings from one to the other...by just moving part of the gsender folder from old to new?

tcurdt, 2025-07-13:
Not sure if this of help but I saw a similar thing. gSender was working fine. I then imported an old config and it got completely stuck at "Connecting..." whether there was a serial port or not.

Moving `~/.config/gSender` away (or deleting it) brought it back into a usable state.

Seems like something in the "Import Config" code is doing something unexpected.

HTH

itsnotlevel1, 2025-07-14:
Ok, im not sure if Scienci will answer this since I only got one reply from a random person and that was appreciated. 

I reinstalled 1.14.12 and reconfigured whatever I needed, even adding my controller buttons back in. I used it for a job and then today I reinstalled 1.5.1 and deleted .sender_rc and sessions folder before the install and start up. I started 1.5.1 and my settings were already there including my game controller. I did not delete any application support files under my username and so far everything is kosher. Is that normal or is this not the normal course of action?

SienciLabs, 2025-07-16:
@itsnotlevel1 it seems like one way or another you've found a way to get 1.5.1 working for your setup

I wish I could update our docs to help others who might find themselves in your situation, but after re-reading this thread several times I'm not exactly sure what steps you took to get it running again for you - if you could offer any insight then I'd appreciate it.

Regarding your other issue where you went to start up Remote mode and it froze gSender for you, this sounds very similar to the other issue #662 so I think if you try to do that again and continue to experience those issues then it'd be good for you to share your experiences there