Sienci-Labs/gsender #658: Spindle Control Issues

gcormier, 2025-06-19:
gSender 1.5.0

Settings
[image]

[gsender-settings-spindle-issues.json](https://github.com/user-attachments/files/20820922/gsender-settings-spindle-issues.json)

Repro
1. Cold boot (gSender + SLBEXT)
2. Connect over IP
3. Turn on spindle

[image]
4. Spindle turns on, yay. Turn off/on all day long.
5. Use dropdown, and choose Durapulse spindle

[image]
6. Turn on spindle. Spindle does not respond, does not spin up.
7. Close gSender. Re-open gSender. Do NOT power cycle SLB-EXT.
8. gSender reverts to "Default Spindle" in the drop down, same as step 3.
9. Turn on spindle. Spindle does NOT respond. Turn "off" spindle.

More info
1. Cold boot SLB-EXT
```
$spindles
0 - SLB_SPINDLE
1 - Huanyang v1
2 - Huanyang P2A
3 - Durapulse GS20, enabled as spindle 0, SDV, current
4 - Yalang YS620
5 - MODVFD
6 - H-100
7 - SLB_LASER, enabled as spindle 1, DIRV
```
2. Choose "Durapulse GS20" from gSender dropdown
```
feeder M104 Q0
ok
feeder $spindles
0 - SLB_SPINDLE
1 - Huanyang v1
2 - Huanyang P2A
3 - Durapulse GS20, enabled as spindle 0, SDV
4 - Yalang YS620
5 - MODVFD
6 - H-100
7 - SLB_LASER, enabled as spindle 1, DIRV
ok
```

There is no longer a "current" spindle active in grblHal ?

gcormier, 2025-06-20:
The key was to define Spindle 2 as GS20.

I think this can be closed, although there might be a UX improvement to be had for gSender detecting this (eg not "running" a spindle that doesn't seem to exist, although I realize it might be more grblHal).

Leaving open for Sienci to decide.

kglovern, 2025-06-24:
We're looking at this, but it's primarily an issue with the firmware that we can't resolve until bringing our version of grblHAL up to date with the current release.  Rest assured, it is in the backlog to verify along when the firmware changes come.

closed 2025-06-24 by kglovern

gcormier, 2025-06-24:
@hamanjam I resolved that via

[image]