Sienci-Labs/gsender #662: Having trouble loading New gsender

sal42na68, 2025-06-20:
Loaded new gsender and everything was fine. I click enable remote for cell phone control and everything froze. I deleted Norton360 and added an exception in windows firewall for gsender. Still can't get past splash screen. I've deleted and re-installed gsender and it still won't run. I deleted new version and installed 1.4.12 and that also gets stuck on splash screen. Anyone know how to fix this?

Thanks

Golfer1234, 2025-06-21:
I'm having same issue?

hamanjam, 2025-06-21:
This was the old "fix". Not sure if it still applies

https://forum.sienci.com/t/how-do-i-revert-versions-in-gsender/9100

sal42na68, 2025-06-22:
Thanks Hamanjam!!! That worked for me. Hopefully they will get the new version fixed.

hamanjam, 2025-06-22:
No problem. I'm using 1.5 and remote is working fine for me.  I don't really use it since I prefer to use windows Remote Desktop but I did test the remote function

Damo21211, 2025-07-21:
@SienciLabs that's okay, went from 1.5.2 down to 1.5. I'll follow that link you sent. Cheers.

sal42na68, 2025-08-07:
New version doesn't freeze but the remote connection to my phone doesn't work. Is there a video for this?

sal42na68, 2025-08-27:
> Since I haven't heard back and we're pretty confident we fixed this issue
> in 1.5.3, I'll close this issue now unless I hear any further response 👍

Doesn't freeze anymore but can't connect Xbox controller