Sienci-Labs/gsender #669: gSENDER 1.5.0 ROTARY SURFACING ISSUES

MattMead166, 2025-06-24:
Why in the rotary surfacing, whether or not the option to "enable rehoming" does the program act with the same results. Shouldn't it spin only one way with enable rehoming checked and multiple ways when not checked? 

Additionally why does each rotary surfacing job get to the end of the stock where it should "be finished" but instead starts spinning the other direction on the A axis and then tries to do a another full pass going back the other way? This is absolutely unneeded, not the expected results and over doubles the surfacing time.

walidkayhan, 2025-06-27:
Based on our testing in v1.5.0, rotary surfacing does not output the same g-code when "Enable Rehoming" is toggled on; the two screenshots below confirm this:

[image]

[image]

Regarding your second question, this also is not the case on v1.5.0, based on our testing.

When rehoming is toggled off, rotary surfacing generates two sets of movements. The first one is for cutting the first half of the stock. Once that is complete, the second set moves the spindle to the end of the stock and cuts the remaining half by moving backwards towards the middle of the stock.

When rehoming is toggled on, there is only one movement, in which the surfacing is done on a single pass.

If you'd like, you can provide me with your rotary surfacing settings, and I can double-check if they cause any incorrect g-code generation that is not doing what I mentioned above :)

MattMead166, 2025-07-09:
I do not have the file saved as it was generated by the rotary surfacing tool. I can absolutely can confirm that the results are the same no matter what is selected. I did four different jobs (all the same size stock) and selected homing and not homing on each one. Each job acted the absolute same continuously spinning in one direction as the x axis worked from left to right, raised z, started spinning the other way continuously and then proceeded to lower z and do another FULL PASS on the x axis from right to left this time even though the surfacing job was already done successfully on each first pass. After the first time I just paused, then stopped each job and returned to home. If I had not stopped them it would of turned each hour job into over two with the unnecessary second passes. I am installing 1.5.1 today and will test the same project again.

SienciLabs, 2025-07-16:
@MattMead166 have you re-tried in 1.5.1 or 1.5.2 yet? If you still see the same results then I'd ask if you could share a picture of you rotary settings that are causing this behaviour.

Same to @walidkayhan when I try this out on my end the two generated g-code files are definitely unique. Also the disabled re-homing definitely cuts halfway, lifts to the end, cuts the other half, then returns to the start - whereas the enabled rehoming spirals the whole way and returns to the start

walidkayhan, 2025-08-11:
@MattMead166 are you still experiencing this issue on the latest version of gSender, v1.5.3?