Sienci-Labs/gsender #673: Toolchange hitting soft limits

gcormier, 2025-06-28:
Possibly related and or regression ? - https://forum.sienci.com/t/gsender-1-2-2-triggers-soft-limits-during-fixed-tool-sensor-change/10712/4

gSender 1.5.1 gets Alarm 2 for soft limits when using tool changes. 

This behavior can be fixed/repro'd simply by adding or removing the M6 T# from the file. The file does not actually need multiple tool changes to trigger anything, it will probe initial tool, next tool, and then throw alarm 2 before even cutting.

This file will trigger it for me.

```
G90 G94
G17
G21
G28 G91 Z0
G90

(First tool. Removing this tool will cause the program to run properly)
T1 M6
S24000 M3
G54
G0 X44.965 Y30.75
Z8

M5
G28 G91 Z0
G28 X0 Y0
G90
M30

```

gcormier, 2025-07-04:
Cannot repo, I will chalk this up to F360 for now.

gcormier, 2025-07-06:
I'm re-opening. I just hit a soft limit using Vectric VCarve Pro. I had thought this was related to some weird side effects of the F360 plugin that will restore rapids and toolchanges to F360, but here we are.

This time

1. Ran a job with 2 tools. Worked fine.
2. Noticed some things I wanted to cleanup on my design. Changed the toolpaths slightly, re-posted with ONLY the 2nd tool.
3. The 2nd tool was still in the machine, so when I ran the file, the initial tool touchoff and subsequent touchoff would be identical results - I suspect this is key.
4. Soft limit alarm thrown - sorry I can't remember the details but I'm guessing it was after I hit "everything looks good, start cutting"

gcormier, 2025-08-12:
I just spent 15 minutes trying to make it happen again, without success. I set my toolchange sensor location to Z0 as well (instead of -10) to see if that would cause things.

Will close, again, and re-open if I can get it to trigger again :)

gcormier, 2025-08-13:
Sorry, I can't go any further back in console to get the first tool info/offsets.

This WAS using the F360 "Post Process All" plugin. @walidkayhan 

```
...
ok
X190.16 Y-279.139 Z-12.678
X190.153 Y-279.136 Z-12.6
ok
G00 Z15.0
G28 G91 Z0
G90
M9
ok
G28 G91 Z0
G90
T1 (M6)
ok
feeder G4 P0.5
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
feeder M5
ok
feeder G4 P0.5
ok
feeder G91 G21
ok
feeder G53 G0 Z-10
ok
feeder G53 G0 X1 Y-1.4
ok
feeder (Toolchange initiated)
ok
feeder G53 G21 G0 Z-1
ok
feeder G53 G21 G0 X331.744 Y-331.744
ok
feeder G53 G21 G0 Z0
ok
feeder (Moving back to configured location)
ok
feeder G90 G53 G0 Z-10
ok
feeder G90 G53 G0 X1 Y-1.4
ok
feeder (This is 10 above configured location)
ok
feeder G53 G0 Z-10
ok
feeder G91 G21
ok
feeder G38.2 Z-158 F400
[PRB:1.000,-1.400,-50.980,0.000:1]
ok
feeder G0 Z2
ok
feeder G38.2 Z-15 F50
[PRB:1.000,-1.400,-50.965,0.000:1]
ok
feeder G21 G10 L20 P0 Z34.715
ok
feeder G0 Z2
ok
feeder G53 G21 G0 Z-10
ok
feeder G21 G91
ok
feeder (Returning to initial position)
ok
feeder G53 G0 Z-10
ok
feeder G90 G21 G0 X190.153 Y-279.137
ok
feeder G90 G21 G0 Z94.59
ALARM:2 (Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked.)
[MSG:Reset to continue]

```

SienciLabs, 2025-08-27:
@gcormier in 1.5.4 we now have some new logic to avoid soft limit triggering from Z movements, could you give this another try and let us know if we've resolved your report?

gcormier, 2025-08-29:
Just got it again.

Was using F360 Post Process All addin. However, I had 5 tool changes. This only happened on the last tool change.

Of note is I went from a VERY short tool to a rather long tool during this change.

The TLO measurement happens. When I click resume, that's when I get the alarm.

```
ok
X29.281 Y54.492 Z0.07 I-0.5 J0
ok
X30.281 Y54.492 Z0.018 I0.5 J0
ok
X30.021 Y54.931 Z0 I-0.5 J0
ok
X29.541 Y54.054 I-0.24 J-0.439
ok
X30.021 Y54.931 I0.24 J0.439
ok
X29.749 Y54.851 I-0.096 J-0.175
ok
G1 X29.701 Y54.764
X29.677 Y54.718 Z0.007
ok
X29.653 Y54.676 Z0.027
ok
X29.634 Y54.64 Z0.059
X29.618 Y54.612 Z0.1
ok
X29.609 Y54.594 Z0.148
ok
X29.605 Y54.588 Z0.2
G00 Z10.675
ok
X29.981 Y54.292
G01 Z2.0 F700.0
Z0.2
ok
G18 G2 X30.181 Z0 I0.2 K0
ok
G1 X30.281
ok
G17 G3 X30.481 Y54.492 I0 J0.2
X29.081 Y54.492 I-0.7 J0
ok
X30.481 Y54.492 I0.7 J0
ok
X30.281 Y54.692 I-0.2 J0
G1 X30.181
ok
G18 G3 X29.981 Z0.2 I0 K0.2
ok
G00 Z20.675
G17
G28 G91 Z0
G90
M9
ok
G28 G91 Z0
G90
T1 (M6)
ok
ok
ok
feeder G4 P0.5
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
feeder M5
ok
feeder G4 P0.5
ok
feeder G91 G21
ok
feeder G53 G0 Z-10
ok
feeder G53 G0 X1 Y-1.4
ok
feeder (Toolchange initiated)
ok
feeder G53 G21 G0 Z-1
ok
feeder G53 G21 G0 X331.744 Y-331.744
ok
feeder G53 G21 G0 Z0
ok
feeder (Moving back to configured location)
ok
feeder G90 G53 G0 Z-10
ok
feeder G90 G53 G0 X1 Y-1.4
ok
feeder (This is 10 above configured location)
ok
feeder G53 G0 Z-10
ok
feeder G91 G21
ok
feeder G38.2 Z-158 F300
[PRB:1.000,-1.400,-51.425,0.000:1]
ok
feeder G0 Z2
ok
feeder G38.2 Z-15 F40
[PRB:1.000,-1.400,-51.415,0.000:1]
ok
feeder G21 G10 L20 P0 Z35.160
ok
feeder G0 Z2
ok
feeder G53 G21 G0 Z-10
ok
feeder G21 G91
ok
feeder (Returning to initial position)
```

[image]

gcormier, 2025-09-03:
Happened again - middle tool in a job.

gcormier, 2026-04-06:
Did this make it into Edge5? Still present in 1.5.7