Sienci-Labs/gsender #682: Gcode error when jogging causing damage.

Tazzoo67, 2025-07-10:
When jogging Z, random corrupt lines / data causing machine to hit Work piece / bed 

[Gsender error.docx](https://github.com/user-attachments/files/21166634/Gsender.error.docx)

[gSender-firmware-settings-7_10_2025-4_33_42 PM.json](https://github.com/user-attachments/files/21166667/gSender-firmware-settings-7_10_2025-4_33_42.PM.json)

SienciLabs, 2025-07-16:
@Tazzoo67 this is now a setting you can find in 1.5.2 called "Continuous jog delay" which you can find just under the jogging presets

I'd recommend to you to try increasing this value much higher than the default, maybe 600 or 700 or even more if you prefer, then you should hopefully find this issue goes away for you.

Let us know how it goes 👍

Tazzoo67, 2025-07-18:
Hi,
    Thanks for the update, I have now installed 1.5.2 and increased the
setting to 700, since these changes have been made I have not seen the
issue occuring so fingers crossed this has fixed it.

Regards,
Dave.

k1jps, 2025-07-18:
> Ok, I think then I'll close this issue as solved for now then! If at any point you have more to add or think the fix didn't fully fix it then you can always come back to this issue to write a new comment and I can re-open it 👍

Thank you for taking care of this problem. I'm in NB Canada on vacation, I'll test it when I get back home.

Capt. JP Savoie
US Coast Guard
License Captain