Sienci-Labs/gsender #686: Gsender froze after clicking remote control

Damo21211, 2025-07-16:
okay guys first up never used the software and seems wicked cant wait to start!!

i was playin around last night got gsender downloaded and got the yorahome 3018 pro workin movin havin fun. 

i select the remote control thingy top of the program and then scanned QR with me phone and then the screen froze and couldnt reopen gsender or anything. checked out the help page and i have gathered the log file 

[2025-07-15 21:39:01.465] [info]  Returned - http://127.0.0.1:57758
[2025-07-15 22:06:02.583] [info]  Returned - http://:::8000
[2025-07-15 22:07:16.888] [info]  Returned - http://:::8000
[2025-07-15 22:13:10.359] [info]  Returned - http://:::8000
[2025-07-15 22:21:21.414] [info]  Returned - http://:::8000
[2025-07-15 23:03:35.676] [info]  Returned - http://:::8000
[2025-07-15 23:09:27.514] [info]  Returned - http://:::8000
[2025-07-15 23:27:57.895] [info]  Returned - http://:::8000
[2025-07-16 19:41:06.266] [info]  Returned - http://:::8000

[image]

is there something i can do to get it running again? im not the most tec savy person only know the basic stuff

Damo21211, 2025-07-16:
It was 1.5.1 cause I tried installing the lasted version 1.5.2 and it doesn't even open the application the computer just ignores me hahah

SienciLabs, 2025-07-16:
Ok, thanks for the info. I think I'll merge this issue with #662 as a duplicate, but it seems it's still an active issue so if you can add any other further information over there then I'd appreciate it.

We'll look into it

closed 2025-07-16 by SienciLabs