Sienci-Labs/gsender #694: individual jog settings for each axis v1.5.2

stefan63, 2025-07-26:
My custom build have a different gearing on Z (using lead screw) compared to XY (beltdrive) that is why I suggest a change for jog presets so that Z can have its own presets.

Kind regards/Stefan

hamanjam, 2025-07-26:
You can adjust both the distance for each axis and the steps.

[image]

[image]

stefan63, 2025-07-27:
That is the main settings for axis I might not have made it clear but I ment the jogging presets.

stefan63, 2025-07-27:
This is what I'm looking for (an ilustration might be better to explain with!

[image]

stefan63, 2025-07-28:
I have done that but the jog preset seams to override this :(
"Thanks for the input @stefan63 but we probably won't do this. For your setup I'd recommend instead adjusting your Z-axis acceleration and velocity values to make the Z-axis movement feel better for you."