Sienci-Labs/gsender #696: 1.5.2 Can't Invert A motor pin

gcormier, 2025-07-26:
Vortex guide asks to invert A enable pin. I don't see this option in gSender 1.5.2

[image]

kglovern, 2025-07-28:
We've added a check for no AXS parse to make sure we always have requested and sucesfully stored that info which should hopefully address this from happening in the future.

gcormier, 2025-07-28:
Awesome - will get the latest build later this evening and report back.

gcormier, 2025-07-28:
Yup, build https://github.com/Sienci-Labs/gsender/actions/runs/16578826756 resolves the issue. Cold boot and gSender start, unlock and home, A axis settings appear and reflect my values without issuing a manual $I.

closed 2025-07-29 by kglovern