Sienci-Labs/gsender #697: 1.5.2 - Diagonal Jogs

gcormier, 2025-07-26:
The 2 lower diagonal jogs go in the wrong direction. The top 2 work correctly.

[image]

gcormier, 2025-08-12:
Yes, machine jogs up/down/left/right correctly, and all my jobs are fine.

SienciLabs, 2025-08-12:
When you do the jogging, in the console does it look like the correct directions are being sent?

gcormier, 2025-08-12:
Yes, the correct commands are sent

https://imgur.com/a/Bg2mY3F

gcormier, 2025-08-12:
I just reset my Y max travel, and toggled rotary axis on and off ($376), and now the diagonal jogs work int he correct orientation. Here's the console output from that

```
ok
feeder $131=648
[MSG:Homing cycle required]
ok
feeder $$
$0=5.0 (Step pulse time, μs)
$1=255 (Step idle delay, ms)
$2=0 (Step pulse invert, mask)
$3=6 (Step direction invert, mask)
$4=8 (Invert step enable pin, boolean)
$5=15 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$8=0
$9=1
$10=511 (Status report options, mask)
$11=0.010 (Junction deviation, mm)
$12=0.002 (Arc tolerance, mm)
$13=0 (Report in inches, boolean)
$14=14
$15=0
$16=0
$17=0
$18=0
$19=0
$20=1 (Soft limits enable, boolean)
$21=1 (Hard limits enable, mask)
$22=15 (Homing cycle enable, mask)
$23=1 (Homing direction invert, mask)
$24=150.0 (Homing locate feed rate, mm/min)
$25=4300.0 (Homing search seek rate, mm/min)
$26=25 (Homing switch debounce delay, ms)
$27=1.500 (Homing switch pull-off distance, mm)
$28=0.100 (G73 pull-off distance, mm)
$29=0.0 (Step Pulse Delay, μs)
$30=24000.000 (Maximum spindle speed, rpm)
$31=5000.000 (Spindle Mode, )
$32=0 (Laser-mode enabled as spindle, boolean)
$33=1000.0 (Spindle Frequency, Hz)
$34=0.0 (Spindle Duty Cycle, %)
$35=0.0 (Minimum spindle speed, %)
$36=100.0 (Maximum spindle speed, %)
$37=8 (Deenergized Steppers, %)
$39=1 (Printable Command Characters, boolean)
$40=1 (Soft Limits Jogging, boolean)
$41=1
$42=2
$43=1 (Homing Locate Cycle, )
$44=4 (Axis Mask, )
$45=3 (Axis Mask, )
$46=0 (Axis Mask, )
$47=0 (Axis Mask, )
$56=5.0
$57=500.0
$58=-5.0
$59=500.0
$60=1 (Restore Default Overrides, boolean)
$61=3 (Ignore Safety Door Signal, boolean)
$62=0 (Sleep Function, boolean)
$63=3 (Disable Laser on Hold, boolean)
$64=0 (Alarm on Startup, boolean)
$65=0 (Allow Feedrate Override, boolean)
$70=11 (Network Service, )
$100=320.000 (X-axis travel resolution, step/mm)
$101=320.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=79.012
$110=15000.000 (X-axis maximum rate, mm/min)
[... 370 lines cut ...]
$511=7
$512=8
$513=8
$520=0
$521=0
$534=0
$664=45
$665=1
$666=0
$668=1
$730=1000.000
$731=0.000
$733=1000.0
$734=0.0
$735=0.0
$736=100.0
$741=3.700
$742=-99.660
$743=0
$744=15
$745=0
ok
feeder $J=G21G91 X3000 F2500
ok
```