Sienci-Labs/gsender #699: Feature - Do not reset WCS on M2

gcormier, 2025-07-29:
I believe GCodeProcessor.ts sets `activeCoordSys=0` when a job ends with M2.

Could this behavior be removed, or, moved to a preference/option that a user can configure?

If I'm working on a piece of wood in a certain location, I don't want to run a job, make some changes, and re-run a job and find it's now working back in WCS1.

gcormier, 2025-07-29:
Looks like M2 is "supposed" to reset the WCS. So I wonder if this could be a feature instead to choose to disable this behavior?

hamanjam, 2025-07-29:
Kevin's link above solved the issue for me a couple years ago.  I have 4 separate workspaces set so I carve some things in one spot while resin cures in another.

closed 2025-07-29 by gcormier