Sienci-Labs/gsender #703: Stuck in Connecting

tcurdt, 2025-08-08:
Starting gSender 1.5.3 on very fast machine (M1 Pro 32GB RAM macOS 15.5) takes forever to start btw.

Then, no matter whether I have a controller connected or not, the program is stuck in "Connecting....".
The worst thing: There seems to be no way to abort.

I have had the problem before on Linux. Same happened now on macOS.
Only after deleting (some of) the config files it comes back to live.
(Yay - reconfigure from scratch! Not great)

Does the config migration have a bug?

Additional but maybe related issues:

1. I've also see it connect (Connect to CNC) but saying "Disconnected" on the top before.

2. The serial ports found have messed up names. Here what they should look like from CoolTerm

[image]

and what gSender shows

[image]

3. Starting gSender from the cli I see

```
/Applications/gSender.app/Contents/MacOS/gSender
2025-08-08T18:10:04.225Z - info init Loading configuration from "/Users/tcurdt/.sender_rc"
(node:69589) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `gSender --trace-warnings ...` to show where the warning was created)
2025-08-08T18:10:04.430Z - info init Starting the server at http://127.0.0.1:64401
20:10:04.430 › Returned - http://127.0.0.1:64401
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
Error: ENOENT: no such file or directory, open '/Applications/gSender.app/Contents/Resources/dist/gsender/app/index.html'
    at async open (node:internal/fs/promises:639:25)
    at async Object.readFile (node:internal/fs/promises:1246:14)
Checking for update
Error: TypeError: Invalid Version: test_error
    at new SemVer (/Applications/gSender.app/Contents/Resources/app/node_modules/semver/classes/semver.js:40:13)
    at compare (/Applications/gSender.app/Contents/Resources/app/node_modules/semver/functions/compare.js:5:32)
    at Object.lt (/Applications/gSender.app/Contents/Resources/app/node_modules/semver/functions/lt.js:4:29)
    at computeReleaseNotes (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/providers/GitHubProvider.js:181:20)
    at GitHubProvider.getLatestVersion (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/providers/GitHubProvider.js:131:35)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async MacUpdater.getUpdateInfoAndProvider (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/AppUpdater.js:319:19)
    at async MacUpdater.doCheckForUpdates (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/AppUpdater.js:333:24)
TypeError: Invalid Version: test_error
    at new SemVer (/Applications/gSender.app/Contents/Resources/app/node_modules/semver/classes/semver.js:40:13)
    at compare (/Applications/gSender.app/Contents/Resources/app/node_modules/semver/functions/compare.js:5:32)
    at Object.lt (/Applications/gSender.app/Contents/Resources/app/node_modules/semver/functions/lt.js:4:29)
    at computeReleaseNotes (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/providers/GitHubProvider.js:181:20)
    at GitHubProvider.getLatestVersion (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/providers/GitHubProvider.js:131:35)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async MacUpdater.getUpdateInfoAndProvider (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/AppUpdater.js:319:19)
    at async MacUpdater.doCheckForUpdates (/Applications/gSender.app/Contents/Resources/app/node_modules/electron-updater/out/AppUpdater.js:333:24)
```

walidkayhan, 2025-08-11:
That is an interesting bug; we can see the command being sent and acknowledged in the console, but the UI is not reflecting any changes to the machine position. Do you mind attaching your diagnostics zip here so we can investigate it further?

walidkayhan, 2025-08-11:
[image]

It's quite easy, actually, just go to the stats page on gSender and click on the "Download Diagnostics File" button, and it should generate a zip file containing some very useful debugging information for us. Please upload the entire zip

tcurdt, 2025-08-11:
I have a few more days while I near the machine.
Let me know what else I can try to help.

[diagnostics_11-08-2025_23-34-48.zip](https://github.com/user-attachments/files/21723489/diagnostics_11-08-2025_23-34-48.zip)

sophiabeluli, 2025-08-25:
I'm glad your connection issue is fixed.

As for the port listings, we have to truncate them as they're too long to fit (especially on smaller screen sizes). In the future, we will be adding named profiles for machines which will change what the port listing names are. That should solve the problem.

SienciLabs, 2025-08-27:
@tcurdt code changes should be in 1.5.4 that's released now 👍 thanks for the testing and feedback!

closed 2025-08-27 by SienciLabs