Sienci-Labs/gsender #708: Probing diameter is forgotten

gcormier, 2025-08-14:
This one took a while to figure out. gSender 1.5.3

repro

1. Choose a probe size (eg 2mm)
2. Probe your part in X
3. Notice the offset
4. Go to settings, and adjust XY thickness from 0 to, say, 0.01mm
[image]
5. Go back to main screen, and probe again. Notice the massive change!
6. To fix - choose a different probe diameter, then re-select your 2mm probe diameter
7. Probe and notice correct behaviour, now offset by value applied in 0.01mm

gcormier, 2025-08-27:
Correct, when selecting a size.

Confirmed still a bug on 1.5.4

gcormier, 2025-09-03:
Latest build artifact resolves this issue. Nice work @sophiabeluli !

SienciLabs, 2025-09-04:
Ok we'll consider it, luckily your approach of using a more recent one is also still a valid approach