Sienci-Labs/gsender #714: Issue: Huge freezes/delays on UI

tomaszglinski, 2025-08-22:
Hi there, I use gSender successfully connected via cable to my FNC ESP32. The only issue I have with UI is that with my files (>100k of lines), during the job it gets unresponsive. So I get an update on the progress literally every few minutes. I cannot reliably stop/pause job, because it will react to my command after minutes. The UI stays freezed even after the job finished.

Just to make it clear - the job itself runs perfectly, streaming goes uninterrupted, so the issue is not a blocker, but extremely irritating.

SienciLabs, 2025-08-27:
@tomaszglinski how large are the files you're loading up and running? We typically run test files up to about 60 or even 80MB but past that the files start to get quite unruly even for more streamlines senders. You might also want to take a look at your systems RAM usage since you might be getting bottlenecked there

tomaszglinski, 2025-08-28:
Hi @SienciLabs, size-wise files rather do not exceed the 50MB. I will check the RAM next time.

tomaszglinski, 2025-09-19:
I just checked, I'm now running the 1.4.12 version since it seemd to me some time ago more stable and it uses now without loaded file 1GB ram. It is possible that the 4Gb were on the 1.5.1 or so. I will install newest and check again. In 1.4.12 I see no diagnostics to download.