Sienci-Labs/gsender #715: Issue: Remaining time calculation completely wrong

tomaszglinski, 2025-08-22:
I have no idea how you calculate the time left, but in my case it always shows few minutes for a job, that has been running for 2 hours and I know it will still run for next few... I use big files (>100k lines) with G04 WAIT commands. Just those G04 can take 2 hours, so I guess you don't take them into consideration?

droric, 2025-08-28:
I ran into this yesterday on a standard vCarve gcode file.  This was during the finishing pass on a carve and the time remaining was approximately 32 minutes when the job completed.  I do not believe there were any additional waits added unless vCarve does that automatically.

Clock56, 2026-10-01:
I am seeing the actual run time around 1.95 x the predicted. I suspect that Gsender does not take account of acceleration/deceleration of the actual machine. I have a python script that reads each gcode movement from the cgode file, it calculates the distance and requested feed, then allows for:
- Each axis’s speed limit.
- Slowing at changes of direction.
- Enough distance to accelerate and brake, looking ahead through subsequent movements.
It adds those movement times together, the result is pretty good.
Just wondering if a furtue release could read the connected CNC's parameters and then do a similar calculation? Other than that a great product, well done.