Sienci-Labs/gsender #718: Manual jogging in gsender 1.5.3 doesn't work well.

zoro186, 2025-08-26:
Hi,
I'm doing the first tests on my homemade CNC (currently nearing completion).
I installed a "PicoCNC with grblHAL" CNC control board and "gSender v1.4.12 on Raspberry PI4 model B (os.ver. Debian GNU/Linux 12 (bookworm)" control software, 
and it seems to be working properly.
Homing isn't activated yet because I need to complete the mechanical transmission part of the Y-axis (I need to get the correct pulleys).
The following problem occurred with gSender 1.5.3 on a Raspberry Pi 4:
Connection to PicoCnc worked without problems, but when manually jogging, the coordinates on the DRO didn't change at all
(or sometimes they did, but with nonsensical numbers, such as: with a 5mm feed command, it indicates a movement of 0.12, the next 0.26, etc.).
The 3D milling symbol doesn't move, but the X, Y, and Z motors move correctly.
Manually, I can't zero the coordinates; the probe doesn't work, but the spindle starts up correctly.
After loading and starting a gcode program, everything started working correctly.
Stopping the gcode program and manually jogging again or zeroing the coordinates made the DRO indicator work correctly.
The same problems occurred when running gSender 1.5.3 on a Windows 11 laptop, but exiting and re-entering the program caused the problem to recur.
While running it with a ArduinoUno r3 works fine without any issues.
Even with ioSender on a Windows 11 notebook, everything works fine.

thnk for attention
Zorko

zoro186, 2025-08-29:
Additional problems with PicoCNC and gSender 1.5.3 with Windows 11
Continuing to test gSender, I encountered other problems listed below:
Starting/stopping the spindle from the console with M3 Sxxxx, M5 actually works, but the rpm reading always says "undefined."
Starting the spindle from the FORWARD key actually starts the spindle, the STOP key stops it, but the FORWARD key indicates that it is always running.
The console is frozen, and gSender must be restarted to restore everything.

walidkayhan, 2025-09-19:
Glad to hear that! Really appreciate the nice comments on gSender, we definitely put a lot of effort into making it a great piece of software to use, happy to hear that others think that as well.

I'll close this issue for now, but if you reencounter this jogging problem, feel free to open it back up, or you can open a new issue :)

closed 2025-09-19 by walidkayhan