Sienci-Labs/gsender #723: 1.5.x is miscalculating the zero position on set tool diameters using the Auto Zero Touch plate.

ArciSupport, 2025-08-30:
Every version after 1.4.11 is setting the zero position of the bit as tangential to the x and y planes instead of the center of the bit when specifying a bit diameter and not using autio or tip.

kglovern, 2025-09-06:
This should be fixed with a recent push and will be part of the next release.

ryansorensen, 2025-09-27:
This is still happening in version 1.5.4 on Mac.

ryansorensen, 2025-09-30:
Sorry for adding to the confusion with the PR. I finally figured out how to run the debug build properly and I can confirm that this is working properly in 1.5.5. Thank you for all your hard work on this awesome app.

walidkayhan, 2025-10-06:
@ArciSupport has this been fixed for you on the latest version of gSender, v1.5.5?

walidkayhan, 2025-10-06:
Glad to hear that, will be closing this issue in that case :)

closed 2025-10-06 by walidkayhan